pepe2k/u-boot_mod #194: unable to reproduce the image

tchen61, 2018-03-10:
i tried building the u-boot image...  (alfa-network-hornet-ub)
following the instruction, i was able to create a binary image... 
the size match the image in original_u-boot_images
alfa-network-hornet-ub_23052014

but the files are different (i did a diff and they are different)

i then loaded it in a NOR flash chip, but it does not seem to boot up at all

one thing i noticed while building was some complaints about
(GIT_DISCOVERY_ACROSS_FILESYSTEM not set, fatal,: not a git repository)
but the make continued and produced

bin/u-boot_mod__alfa-network_hornet-ub__20180309_-git_unknown-unknown.bin

same size as whats in oroginal_u-boot_images but does not work (whatever was in original_u-boot_images does work)

what could be the problem ??

pepe2k, 2018-03-10:
Hi @tchen61,

What toolchain did you use to build the image? How did you download/clone the repository, does it contain hidden .git folder? Do you have git installed on your PC?

Cheers,
Piotr

tchen61, 2018-03-10:
i downloaded the zip and expanded it.... (did not use git clone)

i modified the Makefile and pointing the toolchain to my openwrt staging_dir

export
TOOLPATH=/home/AP_AR9331/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2
#export SUB_MAKE_CMD = $(MAKE) --silent --no-print-directory ARCH=mips V=1
SHELL=$(SHELL)
export SUB_MAKE_CMD = $(MAKE) --silent ARCH=mips
CROSS_COMPILE:=mips-openwrt-linux-uclibc-
export PATH:=$(TOOLPATH)/bin:${PATH}
export CONFIG_BOOTDELAY=1

the make worked, but it does not work... however if i downloaded the
prebuilt original_u-boot_images, then it works fine

the target i am trying is ALFA HORNET UB

so my built was

make  alfa-network_hornet-ub

tchen61, 2018-03-10:
> cell: +1.240.401.7902 <(240)%20401-7902>

what is the best way to debug u-boot without jtag ???

pepe2k, 2018-03-10:
@tchen61,

> i modified the Makefile and pointing the toolchain to my openwrt staging_dir

Try with with current OpenWrt toolchain or show me logs from your build.

> what is the best way to debug u-boot without jtag ???

It depends on what you want to debug.

Cheers,
Piotr

tchen61, 2018-03-10:
here is build log and the Makefile and the resulting alfa-network_hornet-ub
uboot

pepe2k, 2018-03-11:
> here is build log and the Makefile and the resulting alfa-network_hornet-ub

You didn't include it.

Cheers,
Piotr

tchen61, 2018-03-11:
maybe the attachment for this email addr is disabled...???

here it is again

===
vi bl

pepe2k, 2018-03-11:
> maybe the attachment for this email addr is disabled...???

Yes, do it in browser.

Cheers,
Piotr

tchen61, 2018-03-11:
here are the files
1.  build log
https://www.dropbox.com/s/h8cip7g667ptclj/bld.log?dl=0
2. Makefile
https://www.dropbox.com/s/m1o6h9pzqdihy4g/Makefile?dl=0
3. resulting binary... that does not boot
https://www.dropbox.com/s/psf8hf3nx00kf6p/ub.bin?dl=0

thanks
tom

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

tchen61, 2018-03-13:
hello... just curious whether you were able to find  what  my mistake 
was with U-BOOT-MOD ???

in case you did not get the file...

here are the files
1.  build log
https://www.dropbox.com/s/h8cip7g667ptclj/bld.log?dl=0
2. Makefile
https://www.dropbox.com/s/m1o6h9pzqdihy4g/Makefile?dl=0
3. resulting binary... that does not boot
https://www.dropbox.com/s/psf8hf3nx00kf6p/ub.bin?dl=0

thanks
tom

closed 2022-10-13 by pepe2k