pepe2k/u-boot_mod #196: qca_dram cas latency

j-d-r, 2018-04-08:
Hi @pepe2k,

There might be a problem with dram cas, are you sure of your formula ?

2 * mem_cas and high bit in position 31 in function qca_dram_set_ddr_cfg().

If I extract values from QCA u-boot in init953x, init955x and init956x, in CFG_DDR2_CONFIG_VAL the values are:

CAS | MSB | 3LSB
-- | -- | --
3 | 0 | 7
4 | 1 | 1
5 | 1 | 4
6 | 1 | 6

Regards,
Julien

pepe2k, 2018-04-08:
Hi @j-d-r,

With QCA you just can't be sure... Have a look at https://github.com/pepe2k/u-boot_mod/pull/120. So far, I wasn't able to reproduce issue mentioned there and never had any issues with other boards.

Regards,
Piotr

j-d-r, 2018-04-08:
Ok, thank you. That's a mess, and without clear documentation...

I've done it with an \#ifdef and comparisons. (https://github.com/j-d-r/u-boot-QCA956x/commit/b7b21f5052aa2fda256384fe36ea4d9be43117bb#diff-a22fd116cd714c4202b4591f9645e2deR557)

pepe2k, 2018-04-09:
> Ok, thank you. That's a mess, and without clear documentation...
> /* 2 * MEM_CAS is not working on QCA956x */

Does your board hang or doesn't boot at all with default value (2 * MEM_CAS)?

j-d-r, 2018-04-09:
It does not boot at all. Nothing on UART.

CAS should be 5, but if i set  CONFIG_BOARD_DRAM_CAS_LATENCY to 6 (same value in QCA_DDR_CFG_REG than QCA u-boot with 5) it also doesn't boot, maybe because it doesn't match QCA_DDR_CFG2_REG.

But it does not hang completely, I can still set QCA_DDR_CFG_REG to a good value just after calling qdca_dram_init()

gfx18, 2018-09-17:
Hi,
I had problems too. My cpu qca 9533 650 Mhz, tp link tl-wr841hpv3.2, After installing your pepe2k uboot my cpu was running at 400 Mhz, I think it's something to do with Memory cas timing.  I needed your version of u boot to restore my art partition as it needed to be reinstalled after upgrade to 16 Mb flash from 8 Mb.  This router got 64 Mb DDR2 ram.
But I got back to Original/Factory u-boot and my cpu now running at 650 Mhz. My router Hardware details,
GD 25Q64CSIG, 64MBit SPI Flash, 8Mb, SOIC-8
Changed to ,
GD 25Q128CSIG, 128MBit SPI Flash, 16Mb, SOIC-8
Memory: DDR2 SDRAM 800 Mhz : Zentel
512Mb	A3R12E40DBF	32M x 16	1.8V	DDR2-800/1066	FBGA-84	MP
https://imgur.com/a/mHNklm6
SOC : Qualcomm Atheros QCA9533-BL3A 650 Mhz CPU
https://imgur.com/a/6t4ly4e

pepe2k, 2018-09-17:
Hi @gfx18,

> After installing your pepe2k uboot my cpu was running at 400 Mhz, I think it's something to do with Memory cas timing.

This has nothing to do with CAS settings. You should check your reset button definition - GPIO number and polarization. My modification has a recovery mode in case of wrong clock/PLL configuration - if you press reset button on power up, safe clocks (400/400/200) will be applied.

> I'm using openwrt_tl-841n-v10, and it got ddr ram not ddr2.

Your device is different, you shouldn't use U-Boot dedicated for a different model.

Cheers,
Piotr

gfx18, 2018-09-17:
OK, thanks for the info.  But I did install your version of uboot to several others i.e. tl840, tl802 etc. running successfully. They got same SOC!!
I did not have to recover anything, it always booted successfully, may be I'm lucky!!
Thanks anyway, without it I would not be able to update the flash chip.  so Thank you for your version of u-boot!!

pepe2k, 2018-09-17:
> OK, thanks for the info. But I did install your version of uboot to several others i.e. tl840, tl802 etc. running successfully. They got same SOC!!

But they may differ in GPIO (and/or other) configuration.

>  I did not have to recover anything, it always booted successfully, may be I'm lucky!!

Or your device has a similar "issue" as some of the YunCore stuff, have a look:
https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/cpu/mips/ar7240/qca_gpio_init.S#L807

Eventually, it just ignored pressed button and booted the system... just 10s later:
https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/env_scripts.h#L166

>  Thanks anyway, without it I would not be able to update the flash chip. so Thank you for your version of u-boot!!

Welcome!

Cheers,
Piotr

closed 2022-10-13 by pepe2k