pepe2k/u-boot_mod #200: Support 32M Flash?

zhaojh329, 2018-05-04:
Support 32M Flash?

pepe2k, 2018-05-04:
Yes, it's ready in my local tree:

```
u-boot> fli

     Bank: #1
   SPI CS: #0
    Model: Winbond W25Q256
 JEDEC ID: 0xEF4019
     Size: 32 MB (0x2000000)
  Sectors: 8192 x 4 kB (0x1000)
           1024 x 32 kB (0x8000)
           512 x 64 kB (0x10000)
  Address: 0x0 ~ 0x1FFFFFF
     Page: 256 B (0x100)

     Bank: #2
   SPI CS: #2
    Model: Spansion unknown model
 JEDEC ID: 0x010215
     Size: 4 MB (0x400000) (detected)
  Sectors: 64 x 64 kB (0x10000)
  Address: 0x2000000 ~ 0x23FFFFF
     Page: 256 B (0x100)

    Total: 36 MB (0x2400000)
```

sombree, 2018-06-26:
Any chance for pushing those changes to github in near future? :)

closed 2018-06-28 by zhaojh329

ycsunjane, 2020-05-25:
32Mib flash support is useful now.