pepe2k/u-boot_mod #205: Kernel Not Starting ..Uboot hang at Starting Kernel (not always)

srvarma-wisilica, 2018-06-27:
I have built u-boot image for GL-iNETAR300 (AR9344 SOC with 16MB Flash and 128 MB DDR2).

U-boot is booting fine, Valid kernel also there. But sometimes Kernel wont boot. 

It works well if Kernel booting is fine.. Repeated reboots tried but Kernel not booting sometimes, U-boot stuck at "Starting Kernel" print.

What might be the issue?

Regards
Sreekanth

diizzyy, 2018-07-05:
I think I'm seeing the same thing unfortunately I don't have serial on my TL-WDR3600 ver 1.1
Most of the time the device boots just fine however randomly it refuses to boot leaving the switch open. It rarely occurs doing a hard reboot, more often during a soft reboot.

mst-ss, 2018-10-03:
I have similar issues, but very frequently U-boot hangs at the very early stage (AR9344 SOC on Compex WPJ344 board and also on many proprietary boards - similar to DB120). I could narrow it down and it looks like there is a problem with DDR memory controller initialization. I tried DB120 U-boot image from Atheros - it is rock solid, never ever failed. 
@pepe2k do you have any ideas?

pepe2k, 2018-10-03:
@mst-ss 

Could you provide me outputs for below commands from a running Atheros and my U-Boot?

```
md.l 0x18000000 1
md.l 0x18000004 1
md.l 0x18000018 1
md.l 0x18000020 1
md.l 0x18000024 1
md.l 0x18000028 1
md.l 0x180000b8 1
md.l 0x180000cc 1
md.l 0x18000108 1
```

Thanks!

mst-ss, 2018-10-03:
@pepe2k 

Thanks for your prompt reply)

Dump of registers:

u-boot_mod:
18000000: 72FEB115    r...
18000004: 9A10C8A8    ....
18000018: 000000FF    ....
18000020: 0000000C    ....
18000024: 0000000C    ....
18000028: 0000000C    ....
180000B8: 00000E6D    ...m
180000CC: 000FFFFF    ....
18000108: 40000140    @..@

Atheros db120:
18000000: c7d48cd0    ....
18000004: 9dd0e6a8    ....
18000018: 000000ff    ....
18000020: 0000000e    ....
18000024: 0000000e    ....
18000028: 0000000e    ....
180000b8: 00000e59    ...Y
180000cc: 000fffff    ....
18000108: 00000040    ...@

NOTE:
I'm using https://github.com/pepe2k/u-boot_mod/commit/1522a406076be6652861bc534764c9d25035aa26#diff-a22fd116cd714c4202b4591f9645e2de commit with changes taken from https://github.com/pepe2k/u-boot_mod/commit/6bcd876a2b13dc9c6db90e15f91754e6c3c2d6d9#diff-a22fd116cd714c4202b4591f9645e2de 

Thank you for your help!

closed 2022-10-13 by pepe2k