pepe2k/u-boot_mod #210: gpio differences compared to orig u-boot

psyborg55, 2018-09-08:
Hi

i notice some differences in gpio directions/values when loaded original tp-link u-boot. can this cause any problems? tl-mr22u is in question.

TP-LINK U-Boot:
```

gpiochip0: GPIOs 0-31, parent: platform/ath79-gpio, ath79-gpio:
 gpio-0   (                    |sysfs               ) in  lo    
 gpio-1   (                    |sysfs               ) in  lo    
 gpio-2   (                    |sysfs               ) out lo    
 gpio-3   (                    |sysfs               ) out hi    
 gpio-4   (                    |sysfs               ) out hi    
 gpio-5   (                    |sysfs               ) out hi    
 gpio-6   (                    |sysfs               ) out lo    
 gpio-7   (                    |sysfs               ) out lo    
 gpio-8   (                    |sysfs               ) in  lo    
 gpio-9   (                    |sysfs               ) in  lo    
 gpio-10  (                    |sysfs               ) out hi    
 gpio-11  (                    |USB power           ) out hi    
 gpio-12  (                    |reset               ) in  lo    
 gpio-13  (                    |tp-link:blue:system ) out hi    
 gpio-14  (                    |sw1                 ) in  lo    
 gpio-15  (                    |sysfs               ) out hi    
 gpio-16  (                    |sw2                 ) in  hi    
 gpio-17  (                    |sysfs               ) out hi    
 gpio-18  (                    |sysfs               ) out lo    
 gpio-19  (                    |sysfs               ) out lo    
 gpio-21  (                    |sysfs               ) out lo    
 gpio-22  (                    |sysfs               ) out lo    
 gpio-23  (                    |sysfs               ) out hi    
 gpio-24  (                    |sysfs               ) out lo    
 gpio-25  (                    |sysfs               ) out lo    
 gpio-26  (                    |sysfs               ) out lo    
 gpio-27  (                    |sysfs               ) out lo    
 gpio-28  (                    |sysfs               ) out lo    
 gpio-29  (                    |sysfs               ) out lo    
 gpio-30  (                    |sysfs               ) out lo    
 gpio-31  (                    |sysfs               ) out lo    

gpiochip1: GPIOs 494-511, parent: platform/qca953x_wmac, ath9k-phy0:
 gpio-494 (                    |sysfs               ) out lo    
 gpio-495 (                    |ath9k-phy0          ) in  lo    
 gpio-496 (                    |sysfs               ) in  lo    
 gpio-497 (                    |sysfs               ) in  lo    
 gpio-498 (                    |sysfs               ) in  hi    
 gpio-499 (                    |sysfs               ) in  hi    
 gpio-500 (                    |sysfs               ) in  hi    
 gpio-501 (                    |sysfs               ) in  hi    
 gpio-502 (                    |sysfs               ) in  hi    
 gpio-503 (                    |sysfs               ) in  hi    
 gpio-504 (                    |sysfs               ) in  hi    
 gpio-505 (                    |sysfs               ) in  hi    
 gpio-506 (                    |sysfs               ) in  hi    
 gpio-507 (                    |sysfs               ) in  hi    
 gpio-508 (                    |sysfs               ) in  hi    
 gpio-509 (                    |sysfs               ) in  hi    
 gpio-510 (                    |sysfs               ) out hi    
 gpio-511 (                    |sysfs               ) in  hi  
```

modified U-Boot:

```
gpiochip0: GPIOs 0-31, parent: platform/ath79-gpio, ath79-gpio:
 gpio-0   (                    |sysfs               ) in  lo    
 gpio-1   (                    |sysfs               ) in  lo    
 gpio-2   (                    |sysfs               ) out lo    
 gpio-3   (                    |sysfs               ) in  lo    
 gpio-4   (                    |sysfs               ) out hi    
[... 92 lines cut ...]

for reference
modified U-Boot barrier breaker image:

```
GPIOs 0-17, ath79:
 gpio-0   (sysfs               ) in  lo
 gpio-1   (sysfs               ) in  lo
 gpio-2   (sysfs               ) in  lo
 gpio-3   (sysfs               ) in  lo
 gpio-4   (sysfs               ) in  hi
 gpio-5   (sysfs               ) in  hi
 gpio-6   (sysfs               ) in  lo
 gpio-7   (sysfs               ) in  lo
 gpio-8   (sysfs               ) in  lo
 gpio-9   (sysfs               ) in  lo
 gpio-10  (sysfs               ) in  hi
 gpio-11  (USB power           ) out hi
 gpio-12  (reset               ) in  lo
 gpio-13  (tp-link:blue:system ) out hi
 gpio-14  (sw1                 ) in  lo
 gpio-15  (sysfs               ) in  lo
 gpio-16  (sw2                 ) in  hi
 gpio-17  (sysfs               ) in  hi
```

pepe2k, 2018-09-10:
Hi @psyborg55,

First of all:
> also after exporting 494-511 pins and trying to cat debug file i get this for every gpio read attempt:

I fail to see how this is related to U-Boot?

Also, are you aware that ath9k GPIOs == SOC GPIOs in case of your hardware? There are no other (controlled by radio) I/O lines in QCA953x. It's just another way to drive *the same I/O* lines, using two different drivers (ath79-gpio vs. ath9k).

Back to your question...
> i notice some differences in gpio directions/values when loaded original tp-link u-boot. can this cause any problems? tl-mr22u is in question.

From what I can see, GPIO15 and GPIO3 have different directions (they are set to output in TP-Link original U-Boot and left as floating/input in u-boot_mod).

I don't setup/touch GPIOs which aren't used/connected to *anything useful* - in this case they are left in default state which is (usually) floating/input.

As I don't have access to your board, it's up to you to look at the PCB, find and measure these GPIOs to check where they are connected to and whether we should change their direction[/value] in code. TP-Link's code is kinda messy, so I wouldn't be surprised that they changed direction[/value] of these GPIOs by mistake but I'm not entirely sure here.

I briefly looked at this picture from OpenWrt Wiki and I would say that GPIO15 is left unconnected (I believe it's pulled low internally as it's used for XTAL value selection) but GPIO3 seems to be connected *somewhere*:
[image]

PS. OpenWrt ath79 target isn't ready (yet) for production use - please, report your ath9k GPIO related problem in https://bugs.openwrt.org

Cheers,
Piotr

psyborg55, 2018-09-10:
> It's just another way to drive the same I/O lines, using two different drivers (ath79-gpio vs. ath9k).

I'm aware of that. Is ath9k gpio driver upstreamed into kernel maybe?

> GPIO3 seems to be connected somewhere:

I'll check that. Meanwhile I found by extracting original firmware these is an ipad_charge binary in /usr/sbin directory. Quick lookup on google reveals that ipad needs charging current of 2.1A. Maybe some gpio passes more current to USB port when toggled?

jack338c, 2021-07-16:
hi
gpio-11 USB power of mr22u may not work ,I tried echo 0 >/sys/class/gpio/gpio11/value to turn off usb power,but it dosen't work.
please help
thanks

psyborg55, 2021-11-27:
there is no way to turn off usb power on this device. it might be possible to rewire EN pin of TPS2553 power switch to a free GPIO but that requires opening device and soldering.

> but GPIO3 seems to be connected _somewhere_:

gpios 0-3 are pulled out to unpopulated header probably used for jtag

recently i had dismantled device to replace dying battery and upgraded memory to 64MB while at it. also fixed the the SPI chip mess and replaced dead pullup/pulldown resistors on its pins.

then the device booted with gpio3 and gpio4 low, tried changing direction/value but it worked only for gpio4

this is current gpio debug output:

```
gpiochip0: GPIOs 0-31, parent: platform/ath79-gpio, ath79-gpio:
 gpio-0   (                    |sysfs               ) in  lo
 gpio-1   (                    |sysfs               ) in  lo
 gpio-2   (                    |sysfs               ) out lo
 gpio-3   (                    |sysfs               ) in  lo
 gpio-4   (                    |sysfs               ) out hi
 gpio-5   (                    |sysfs               ) out hi
 gpio-6   (                    |sysfs               ) out lo
 gpio-7   (                    |sysfs               ) out lo
 gpio-8   (                    |sysfs               ) in  lo
 gpio-9   (                    |sysfs               ) in  lo
 gpio-10  (                    |sysfs               ) out hi
 gpio-11  (                    |sysfs               ) in  lo
 gpio-12  (                    |reset               ) in  lo
 gpio-13  (                    |tp-link:blue:system ) out hi
 gpio-14  (                    |sw1                 ) in  hi
 gpio-15  (                    |sysfs               ) in  lo
 gpio-16  (                    |sw2                 ) in  hi
 gpio-17  (                    |sysfs               ) out hi
```

not sure if it matters but gpio4 (and gpio17) is also connected to holtek MCU. for what purpose i have no idea.
besides the smart power path design flaw, everything else works pretty stable so i guess small gpio differences are not relevant

closed 2021-11-27 by psyborg55

jack338c, 2021-11-27:
@psyborg55 
thank you for your reminder