pepe2k/u-boot_mod #211: always get make: *** No rule to make target 'tplink_wr703n'. Stop

jack338c, 2018-09-10:
Dear
please help me

here are my steps
...........................................
git clone https://github.com/pepe2k/u-boot_mod.git

cd u-boot_mod

wget https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/OpenWrt-Toolchain-ar71xx-generic_gcc-5.3.0_musl-1.1.16.Linux-x86_64.tar.bz2

mkdir toolchain

extrat OpenWrt Toolchain to toolchain folder 
(/home/jack/u-boot_mod/toolchain/OpenWrt-Toolchain-ar71xx-generic_gcc-5.3.0_musl-1.1.16.Linux-x86_64/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16)

##edit /home/jack/u-boot_mod/Makefile

change
...................
export HOSTOS
export HOSTARCH
export BUILD_TOPDIR = $(PWD)
export STAGING_DIR  = $(BUILD_TOPDIR)/tmp
export SOURCE_DIR   = $(BUILD_TOPDIR)/u-boot
export BIN_DIR      = $(BUILD_TOPDIR)/bin
export SUB_MAKE_CMD = $(MAKE) --silent --no-print-directory \
                      ARCH=mips V=1 SHELL=$(SHELL)

to
..............

export HOSTOS
export HOSTARCH
export BUILD_TOPDIR = $(PWD)
export STAGING_DIR  = $(BUILD_TOPDIR)/tmp
export SOURCE_DIR   = $(BUILD_TOPDIR)/u-boot
export BIN_DIR      = $(BUILD_TOPDIR)/bin
export SUB_MAKE_CMD = $(MAKE) --silent --no-print-directory \
                      ARCH=mips V=1 SHELL=$(SHELL)

export MAKECMD=make --silent ARCH=mips CROSS_COMPILE=mips-openwrt-linux-uclibc-
export PATH:=$(BUILD_TOPDIR)/toolchain/bin/:$(PATH)

then run make tplink_wr703n

but always get "make: *** No rule to make target 'tplink_wr703n'.  Stop."

please thelp me 

thank you

pepe2k, 2018-09-10:
Hi @jack338c,

Have a look: https://github.com/pepe2k/u-boot_mod/blob/master/Makefile#L221

Cheers,
Piotr

closed 2018-09-10 by pepe2k

jack338c, 2018-09-29:
Thank you ! 
I will try