pepe2k/u-boot_mod #218: No Qualcomm Atheros QCA955x

montvid, 2018-11-14:
There is no  Qualcomm Atheros QCA955x support but the readme says that there is. Please fix the readme or state what  Qualcomm Atheros QCA955x devices do you support. Thanks.

psyborg55, 2019-07-16:
you can easily support your device based on my PR https://github.com/pepe2k/u-boot_mod/pull/229

montvid, 2019-07-17:
Thanks for the information. My device is TP-Link TL-WR1043ND v2 Qualcomm Atheros QCA9558 ver 1 rev 0. I'm using the new openwrt ath79 build successfully. I don't know how to modify this in order not to brick my device and I don't have the tools to fix it. But thanks for trying to support these devices!

closed 2020-05-31 by pepe2k

montvid, 2020-05-31:
Fix your readme and then close.

`First version of this modification was introduced on OpenWrt forum in this thread, at the end of March 2013 and was dedicated only for TP-Link routers with Atheros AR9331 SoC. Now, it supports also models from different manufacturers, devices with Atheros AR934x, Qualcomm Atheros QCA953x, Qualcomm Atheros QCA955x and other (in the near future Qualcomm Atheros QCA956x and MediaTek MT762x) are under development.`