pepe2k/u-boot_mod #220: 16BM Winbond W25Q128FVSG support?

ohsorry, 2018-12-19:
here i found a [datasheet](http://www.datasheetq.com/w25q128fvsg-doc-winbond)

CodeFetch, 2019-03-04:
@ohsorry This chip is already supported. I have already used it for some routers.

apnof, 2019-03-04:
Can confirm - chip working.

Chip Example: [aliexpress-W25Q128FVSG](https://www.aliexpress.com/item/W25Q128FVSSIG-SOP8-5-2MM-25Q128FVSG-16M/32822722931.html?spm=a2g0s.9042311.0.0.27424c4d7woWrS) (got it for ~50 cents a piece).
Programmer example: Working with this programmer [aliexpress-CH341A](https://www.aliexpress.com/item/Free-shipping-Bios-Board-MX25L6405-W25Q64-USB-Programmer-LCD-Burner-CH341A-Progammer-For-24-25-Series/32283031779.html?spm=a2g0s.9042311.0.0.27424c4d7woWrS) (<2€) in combination with the software [flashrom](https://www.flashrom.org/)

ohsorry, 2019-03-28:
Excellent!

closed 2019-03-28 by ohsorry