pepe2k/u-boot_mod #224: 32 MiB support

CodeFetch, 2019-04-17:
@pepe2k in #200 you said that you have support for dual flash and 32 MiB already in your local tree. Can you please push this?

lePereT, 2020-08-03:
Hey @CodeFetch thanks for this request. Would love to see this work with bigger flash chips. I have one question and that has to do with a comment made by @pepe2k that the address of the ART partition is coded in a mach file in a way that only enables the use of 16MiB of flash - is this barrier still present in DTS based implementations?

CodeFetch, 2020-08-08:
@lePereT I think you're right, but I'm not sure about the consequences. This patch took care: https://github.com/openwrt/openwrt/blob/d3c9630811468499a94302fc57f3a4874f1614bc/target/linux/ath79/patches-4.19/405-mtd-tp-link-partition-parser.patch

Now we only have MTDSPLIT: https://github.com/openwrt/openwrt/blob/master/target/linux/generic/files/drivers/mtd/mtdsplit/mtdsplit_tplink.c

@adrianschmutzler Do you know if the size of the flash or better say master->size is still being automatically detected? If yes, I wonder why the ART-Partition wasn't included as part of the TP-Link flash layout. Is the DTS-hardcoded art-partition used instead? Do you have an idea how to reintroduce automatic adaption to the flash size again?

salexa74, 2021-07-27:
@pepe2k, hi! Could you publish somewhere *.bin or source code "as is" with support of the SPI Flash more than 16M? For TL-WDR4300 I replaced SPI: MX25L25645G (32M/256Mbit), and assembled an firmware image based on OpenWrt 19.07.7 with MTD layout for 32M SPI. Everything works fine, but I would like to be able to flash firmware image >16M by the bootloader directly.

closed 2022-10-13 by pepe2k