pepe2k/u-boot_mod #230: Add Gzip support

ramponis, 2019-07-17:
Is it possibile to add gzip compression to u-boot?
I have some gzip compressed image and i want to test on a TL-WDR3600

Thank you

CodeFetch, 2019-08-11:
No. LZMA is superior to GZIP. What's so interesting about this image?

ramponis, 2019-08-19:
Yes i know LZMA is superior... but i want to test some images compressed with GZIP.
I don't have the sources to recompile the image with LZMA compression

CodeFetch, 2019-08-19:
That's likely a GPL violation. Ask the person you've got the image from where the source-code is available.

As far as I know you can also just extract the image with `binwalk -e` and repack it with OpenWrt's mkimage to LZMA. Please ask in the OpenWrt forum for help for that.

You can also add GZIP support yourself to U-Boot_Mod, but don't expect someone to help you, because GZIP compression was removed by purpose. Not only is GZIP compression inferior, but you would also need the decompressor which increases the size of the U-Boot. 
Look e.g. at this line https://github.com/pepe2k/u-boot_mod/blob/7a540a78df6082ca9cc82f72c6a83d34938c2503/u-boot/common/cmd_bootm.c#L208

You see the GZIP cases where commented everywhere, but you could introduce a new build flag for switching between compression methods.

closed 2020-05-31 by pepe2k