pepe2k/u-boot_mod #231: uboot mod for Huawei B5328

coolirc, 2019-07-29:
Hello i need some help concerning a 4G Lte device which uses uboot i have several devices that are stuck on some sort of a debug mode which telnet is enabled in this mode and i can see all the file system , partitions etc ... unfortunately when the device is working under normal mode the telnet is disabled and i can't stop the uboot from booting the kernel i think the manufacturer has disabled it ... i have only access to console port which are rx tx and gnd. 
here are informations about the device : 

`dev:    size   erasesize  name
mtd0: 00080000 00020000 "u-boot"
mtd1: 00080000 00020000 "env"
mtd2: 00100000 00020000 "rev0"
mtd3: 00100000 00020000 "ltenv"
mtd4: 00100000 00020000 "wmnv"
mtd5: 00100000 00020000 "cmnnv"
mtd6: 00100000 00020000 "cmnnv2"
mtd7: 00400000 00020000 "rev1"
mtd8: 00400000 00020000 "linux"
mtd9: 00400000 00020000 "linux2"
mtd10: 01e00000 00020000 "rootfs"
mtd11: 01e00000 00020000 "rootfs2"
mtd12: 00500000 00020000 "tk"
mtd13: 00500000 00020000 "tk2"
mtd14: 00080000 00020000 "customize"
mtd15: 00280000 00020000 "log"
mtd16: 02000000 00020000 "update"`

`# uname -a
Linux B5328 3.10.0-uc0 #1 Wed Aug 24 09:38:10 CST 2016 armv7b GNU/Linux
# atc ati
sh: atc: command not found
# at
sh: at: command not found
# ls
bin      dev      etc      home     lib      linuxrc  mnt      proc     root     sbin     sys      tmp      usr      var
# mount
rootfs on / type rootfs (rw)
ubi0:rootfs on / type ubifs (ro,relatime)
proc on /proc type proc (rw,relatime)
ram on /var type ramfs (rw,relatime)
sys on /sys type sysfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=666)
/dev/mtdblock15 on /var/log type jffs2 (rw,relatime)
#

# cat /proc/mounts
rootfs / rootfs rw 0 0
ubi0:rootfs / ubifs ro,relatime 0 0
proc /proc proc rw,relatime 0 0
ram /var ramfs rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
devpts /dev/pts devpts rw,relatime,mode=666 0 0
/dev/mtdblock15 /var/log jffs2 rw,relatime 0 0

# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7b)
BogoMIPS        : 1652.73
Features        : swp half fastmult vfp edsp neon vfpv3 tls vfpv4 idiva
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 4

Hardware        : GCT GDM7243
Revision        : 0000
Serial          : 0000000000000000

`
here is the nand dump i could perform from telnet . [dump](http://www.mediafire.com/file/5jjxjof59i57192/nand_dumpB5328.zip/file)

i need to patch the uboot or a way to flash or take the firmware using tftp and uboot method 

thanks for help

coolirc, 2019-07-30:
i was able to make a connection using serial port with rx tx and gnd now i was able to printenv but i am unable to ping to pc using address 192.168.0.10 which i set manually . 
is it because the current env variable **ethact=mii0**
gatewayip=192.168.0.1
netmask=255.255.255.0
ipaddr=192.168.0.1
serverip=192.168.0.10
ethact=mii0
pesifwcheck=1

here is the output of the command printenv : 

`
GDM7243> printenv
baudrate=921600
fs_permission=ro
nand0=gdm7243
mtdparts=mtdparts=gdm7243:512k(u-boot),512k(env),1m(rev0),1m(ltenv),1m(wmnv),1m(cmnnv),1m(cmnnv2),4m(rev1),4m(linux),4m(linux2),30m(rootfs),30m(rootfs2),5m(tk),5m(tk2),512k(customize),2560k(log),32m(update)
kernel_exec_addr=0xd05fffc0
kernel_load_addr=0xd0600000
file_load_addr=0xd0008000
cleanenv=nand erase.part env
linuxdn=loady ${kernel_load_addr}
imagedn=loady ${file_load_addr}
linuxsave=image write ${kernel_load_addr} linux ${filesize}
linux0save=image write_part ${kernel_load_addr} linux ${filesize}
linux2save=image write_part ${kernel_load_addr} linux2 ${filesize}
rootfssave=image write ${file_load_addr} rootfs ${filesize}
rootfs0save=image write_part ${file_load_addr} rootfs ${filesize}
rootfs2save=image write_part ${file_load_addr} rootfs2 ${filesize}
tksave=image write ${file_load_addr} tk ${filesize}
tk0save=image write_part ${file_load_addr} tk ${filesize}
tk0save=image write_part ${file_load_addr} tk2 ${filesize}
linuxsave2=nand erase.part linux;nand write ${kernel_load_addr} linux ${filesize}
rootfssave2=nand erase.part rootfs;nand write.jffs2 ${file_load_addr} rootfs ${filesize}
paramsave=nand erase.part param;nand write ${file_load_addr} param ${filesize}
bootsave=nand erase.part u-boot;nand write ${file_load_addr} u-boot ${filesize}
bswoecc=nand erase.part u-boot;nand write.woecc ${file_load_addr} u-boot ${filesize}
setbootargs=set bootargs console=ttyS0,${baudrate} boot=nand mem=${mem} ubi.mtd=10,2048 root=ubi0:rootfs ${fs_permission} rootfstype=ubifs ${mtdparts}
bootcmd=image read ${kernel_exec_addr} linux;bootm ${kernel_exec_addr}
bootcmd2=run setbootargs;nand read ${kernel_exec_addr} linux;bootm ${kernel_exec_addr}
webupdatekernel=0
webupdaterootfs=0
setfactrst=0
webupdkernelcmd=nand erase.part linux;nand read ${kernel_load_addr} upuimg ${partsize};nand write ${kernel_load_addr} linux ${partsize}
webupdrootfscmd=nand erase.part rootfs;nand read ${file_load_addr} upram ${partsize};nand write ${file_load_addr} rootfs ${partsize}
webupdatecmd=run webupdkernelcmd;run webupdrootfscmd;run bootcmd
factrstcmd=nand erase.part linux;nand read ${kernel_load_addr} factrst0 ${partsize};nand write ${kernel_load_addr} linux ${partsize};nand erase.part rootfs;nand read ${file_load_addr} factrst1 ${partsize};nand write ${file_load_addr} rootfs ${partsize}
upuimgsave=nand erase.part upuimg;nand write ${kernel_load_addr} upuimg ${filesize}
upramsave=nand erase.part upram;nand write.jffs2 ${file_load_addr} upram ${filesize}
factrst0save=nand erase.part factrst0;nand write.jffs2 ${kernel_load_addr} factrst0 ${filesize}
factrst1save=nand erase.part factrst1;nand write.jffs2 ${file_load_addr} factrst1 ${filesize}
tftpuboot=tftp ${file_load_addr} u-boot.bin
tftplinux=tftp ${kernel_load_addr} uImage
tftpcramfs=tftp ${file_load_addr} ramdisk.cramfs
tftpjffs2=tftp ${file_load_addr} ramdisk.jffs2
tftpubifs=tftp ${file_load_addr} ramdisk.ubifs
tftptk=tftp ${file_load_addr} tk.gz
ethaddr=00:0a:3b:f0:01:30
gatewayip=192.168.0.1
netmask=255.255.255.0
ipaddr=192.168.0.1
serverip=192.168.0.10
ethact=mii0
pesifwcheck=1
mtdids=nand0=gdm7243
mem=80M
op_bootargs=auto
bootdelay=5
stdin=serial
stdout=serial
[... 43 lines cut ...]
15: log                 0x00280000      0x05880000      0
16: update              0x02000000      0x05b00000      0

active partition: nand0,0 - (u-boot) 0x00080000 @ 0x00000000

defaults:
mtdids  : nand0=gdm7243
mtdparts: mtdparts=gdm7243:512k(u-boot),512k(env),1m(rev0),1m(ltenv),1m(wmnv),1m(cmnnv),1m(cmnnv2),4m(rev1),4m(linux),4m(linux2),30m(rootfs),30m(rootfs2),5m(tk),5m(tk2),512k(customize),2560k(log),32m(update)
`
i deleted from telnet the partition of the kernel so the device is not able to load the kernel but i have a nand dump of each partition already made before on a working one .
now my question is how can i perform a backup of that partition using uboot ? 

thanks for help 

uboot version is 2010 

`U-Boot 2010.09 (Sep 06 2016 - 10:08:39)GCT GDM7243

Build Info:
  date: 2016/09/06-10:08:44
  user: root@ubuntu-will
  svnr:
  src: /home/will/DEVELOPMENT/LTE_Router/B5328_FDD/SDK/work/uboot
  ver: 0.46e
`

CodeFetch, 2019-08-11:
This is not a forum and we don't offer help for random Huawei devices. Do us a favor and close this issue...

coolirc, 2019-08-13:
Thanks for your reply i'll close the ticket , i'd expected some help from your side but thanks ...

closed 2019-08-13 by coolirc

CodeFetch, 2019-08-13:
@coolirc This is just not the correct place, because this project has nothing to do with Huawei U-Boot and opening an issue in a different software project is not the correct way to ask for help. We don't know anything about Huawei-U-Boots and especially I don't know what you want to achieve. If you want to get a dump of the partitions, try to read it out directly. If that doesn't work, write a script that abuses the md command.

pepe2k, 2019-08-13:
@CodeFetch 

> This is not a forum and we don't offer help for random Huawei devices. Do us a favor and close this issue...

"We"? Why are you speaking on my behalf? This is not your repository and not your project, so next time please be nice for others and speak only on your behalf.

Cheers,
Piotr

CodeFetch, 2019-08-13:
Understood, but "we" was meant as people having to do with ath79 U-Boot. If you ask about motor issues with your yacht in a sports sailboat forum, I guess sailors would refer to themselves as "we", too. And I think it's also not nice to bug people opening an issue/bug report asking for help for a unrelated thing like in a forum.

coolirc, 2019-08-14:
well i'm not totally out of subject like you say , i was hoping to port uboot to that board since it's a new chip and since that repository was named **U-boot_Mod** so it's actually being in the subject .

CodeFetch, 2019-08-14:
@coolirc Ah okay... This U-Boot-Mod is for special WLAN routers with Qualcomm/Atheros chipsets. Your Huawei has a Realtek chipset and as you were already told in OpenWrt forum, chances are bad for easily supporting this device without source-code or proper specifications. 

Write an email to Huawei and ask them where to find the source code and if they don't have it available e.g. on the accompanying CD or on the internet, tell them they need to release the source-code as it's a GPL violation otherwise. This also includes the U-Boot source code (while I doubt that you need a custom U-Boot at all).

Personally I think this device is not worth the effort for trying to write a custom U-Boot or even supporting it in OpenWrt, because it is overpriced. I recommend you to sell it on eBay before it's getting completely outdated and worthless. I have bought a Huawei 4G router 2 years ago for $150 which at that moment costed more than $400 on eBay. Now it sells for under $30.

coolirc, 2019-08-15:
sources : https://github.com/torvalds/linux/tree/master/drivers/staging/gdm724x

CodeFetch, 2019-08-15:
@coolirc That's only the source code for the SoC. Please ask in the OpenWrt forum for help.