pepe2k/u-boot_mod #247: Update u-boot

uis246, 2020-08-18:
1.1.4 from 2013 is too old. Why not to migrate to new version. Or even commit to upstream.

mpratt14, 2020-08-18:
I imagine the largest reason is space limitations. Some boards only have room for 64 KB of lzma image

uis246, 2020-08-18:
Leave this version only for those boards. Or disable unsused features, enable spase optimization and uninlining.

lePereT, 2020-08-24:
Apart from the web recovery, what does this U-Boot have that's not present in mainline? Flash chip detection? At the moment I'm considering whether to use this as the base for a project or mainline...

CodeFetch, 2020-11-25:
My plan is to create a bootloader from scratch sharing common DTS files with Barebox specifically fit for routers. Let me know if you have an idea for a name or want to contribute in any way (code, documentation, funding).

uis246, 2020-11-25:
@CodeFetch Aplan-e

CodeFetch, 2020-11-26:
@uis246 I know it's not good to create a fork or something, but there is no proper base for bootloaders for routers. That's why most are based on an old version of U-Boot. Barebox fits better than U-Boot for routers as most run linux and Barebox is kind of a linux-friendly U-Boot version, but the "Linux features" of Barebox are for big Intel/AMD machines with hard disks and EXT file systems... That's why I'm going to start from scratch, but keep compatibility with Barebox to reduce double-work... Next thing I see with pepe2k's U-Boot mod is that people need to get involved. That's why I want to simplify the code. The entry barrier is to high atm. If your name suggestion was meant serious: I fear people don't get the idea it could be a project for a bootloader for networking equipment.

youxiaojie, 2021-01-14:
> @uis246 I know it's not good to create a fork or something, but there is no proper base for bootloaders for routers. That's why most are based on an old version of U-Boot. Barebox fits better than [...]

support!!

youxiaojie, 2021-01-14:
does your u-boot support ar7242?

closed 2022-10-13 by pepe2k