pepe2k/u-boot_mod #257: Unbrick WR841N after wrong u-boot-flash

davideciarmiello, 2022-05-04:
Hi.

i have flashed wrongly a u-boot :( i have writed a wrong name in the second line! 
[image]

Now the router not start, when i power on it, only power led is on, and Internet led is on (little).

I have connected it with putty over a serial cable, but I don't see anything in console.

is there any solution? or the solution is to throw it away?

Thanks very much :)

mpratt14, 2022-05-04:
something I have done before:

get an identical device, copy the entire flash data to a file (might have to copy each partition and then concatenate them)

getting another device can be skipped if you only want to get uboot on

a cheap serial programmer like a FT232H breakout, to write the file to the flash chip (I had to write my own scripts in Python to do so). or an expensive one which requires no software skills.

there are cables for sale that can clip onto the chip so that no soldering is necessary, however it is more likely to work when desoldering the chip first and then connecting wires to its pinouts

davideciarmiello, 2022-05-07:
OK I think it should be thrown away :)
Thanks for your support!

mpratt14, 2022-05-07:
I would try to find someone to donate it to for free

closed 2022-10-13 by pepe2k