pepe2k/u-boot_mod #258: Firmware uploading is interrupted randomly by some packets

matlinuxer2, 2022-05-12:
## Descriptions:

Model: Tube-E4G (code base could be different than the latest upstream) [Model info](https://openwrt.org/toh/hwdata/alfa_network/alfa_network_tube-e4g)

When I uploaded a firmware smaller than 8MB(ex: [Openwrt Tube-E4G image](https://downloads.openwrt.org/releases/21.02.3/targets/ramips/mt7620/openwrt-21.02.3-ramips-mt7620-alfa-network_tube-e4g-squashfs-sysupgrade.bin) ), the process finished smoothly. But when I tried a larger firmware
(ex: [Alternative Image listed in ALFA docs site](https://docs.alfa.com.tw/Support/OpenWrt/img/openwrt-ramips-mt7620-alfa-network_tube-e4g-squashfs-sysupgrade.bin) , 8.5MB)
It failed often, around 80% failed.

## Collected information:

I captured the packets logs with the wireshark. The log file is 

[2022-05-09_uboot-tube-e4g_analysis.log.pcapng.zip](https://github.com/pepe2k/u-boot_mod/files/8677180/2022-05-09_uboot-tube-e4g_analysis.log.pcapng.zip)

I attach some screenshots for comparison here:

Failed case: 
[image]

Success case:
[image]

It looks like some TCP Dup ACK packets interrupt the process, but don't know what's cause this issue in the background.
In my debug instance, the failed case's process didn't enter the block here:
https://github.com/pepe2k/u-boot_mod/blob/ad5a2135bdba57454c1fb9fe4d8efd7b04d0cfe5/u-boot/httpd/httpd.c#L497

So far these are all clues I can get for now.

matlinuxer2, 2022-05-17:
After trace the issue more deeply, I found the root cause is not TCP Dup ACK but the http request length calculation issue.
In some cases, it could be `hs->upload` is greater than `hs->upload_total` but the http request still not end. The `webfailsafe_post_done` change and `http_state_reset` are trigged before the last packet arrived, which blocks the whole process.

To fix this issue, I add a patch to detect the http request end based on `Content-Length` parameter: https://github.com/matlinuxer2/u-boot_mod/commit/0c655af7bd502f186438ce891a559c95491303aa

I tested it with Tube-E4G and chrome on Linux, not sure it apply to other cases yet. 
Hope it helps

matlinuxer2, 2022-05-18:
Hi @pepe2k 
I attached the tested bootloader below:

[bootloader_binaries.zip](https://github.com/pepe2k/u-boot_mod/files/8720838/bootloader_binaries.zip)

I also split my changes into 3 commits, which could be easier to review and filter debug-only parts, commits are here:

https://github.com/matlinuxer2/u-boot_mod/tree/fix_web_failsafe_stall

The captured logs I upload below:

[success_firefox.zip](https://github.com/pepe2k/u-boot_mod/files/8720858/success_firefox.zip)
[failed_firefox.zip](https://github.com/pepe2k/u-boot_mod/files/8720861/failed_firefox.zip)
[patched_firefox.zip](https://github.com/pepe2k/u-boot_mod/files/8720862/patched_firefox.zip)
[failed_chrome.zip](https://github.com/pepe2k/u-boot_mod/files/8720874/failed_chrome.zip)
[success_chrome.zip](https://github.com/pepe2k/u-boot_mod/files/8720875/success_chrome.zip)
[failed_curl.zip](https://github.com/pepe2k/u-boot_mod/files/8720876/failed_curl.zip)
[success_curl.zip](https://github.com/pepe2k/u-boot_mod/files/8720878/success_curl.zip)

slandden, 2022-08-28:
I am having problems with connection getting reset, with8MB [1] I can get it working, but with official gl.inet images 50MB it always resets.

Slate AX

[1] https://github.com/coolsnowwolf/openwrt-gl-ax1800

closed 2022-10-13 by pepe2k